Reefwing-Software/Reefwing-AHRS #3: yaw and pitch interconnection/tilt-compensation problem.

FugLong, 2023-01-30:
Hi, I'm integrating reefwing_ahrs with the nano 33 ble and opentrack to create a wired/wireless head tracker for PC games, but this library has been giving me some trouble.

Basically no matter what filter type I choose, angles.pitch affects angles.yaw. Yaw and roll work perfectly alone and combined, but even if I disable output from pitch entirely, somewhere in the math pitch is still affecting yaw.

I spent a day or two assuming it was axis misalignment; trying to verify the orientation of the LSM9DS1 IMU's axis and the order and signs needed in the quaternion update calls (or the filter function in this case) but from what I can find your library is compensating properly for the weird mag orientation of the 33 ble. 

Another possibility is that the compass tilt compensation is busted, but this seems unlikely to me because yaw is only affected by one of the other axis not both. 

I have mainly been using Fusion filter. It works the best for this application imo, but all of the filter types are broken in the same way: pitch always affects yaw. 

So basically I'm at a loss of what's wrong or how to fix it, but I'm not smart enough to implement Fusion on my own. Any ideas or tips would be greatly appreciated. *I'm attaching my current ino file, ignore the hat struct and weird serial writes, that is only used when talking with opentrack software. It's currently just outputting strings and ints to serial in test mode*. 

Thanks for this lib by the way!

```
#include <ArduinoBLE.h>
#include <ReefwingAHRS.h>

// LED Pin definitions
// *RGB PINS ARE INVERTED ON THIS BOARD*
// for RED, BLUE, GREEN, and BUILTIN only: HIGH means LOW and vice versa.
#define PIN_LED     (13u)
#define LED_BUILTIN PIN_LED
#define RED        (22u)
#define GREEN        (23u)
#define BLUE        (24u)
#define LED_PWR     (25u)

LSM9DS1 imu;
EulerAngles angles;

//TEST MODE ENABLE OR DISABLE
bool TestMode = true; //Set true to output data directly to serial (bypass hatire conversion + centering)

//Variables only required in test mode
int loopFrequency = 0;
const long displayPeriod = 100;
unsigned long previousMillis = 0;

//Centering variables (currently disabled)
int center_yaw = 0;
int center_pitch = 0;
int center_roll = 0;
int loopCount = 0; //Used for centering once filter is calm

//BLE variables + init
bool BLEconnected = false;
const char* deviceServiceUuid = "19b10000-e8f2-537e-4f6c-d104768a1214";
const char* deviceServiceCharacteristicUuid = "19b10001-e8f2-537e-4f6c-d104768a1214";
const char* deviceServiceCharacteristicUuid1 = "19b10002-e8f2-537e-4f6c-d104768a1214";
BLEService opentrackService(deviceServiceUuid); 
BLECharacteristic hatireCharacteristic(deviceServiceCharacteristicUuid, BLERead | BLENotify, 30, true);
BLECharacteristic writeCharacteristic(deviceServiceCharacteristicUuid1, BLEWrite, 30, false);

//this structure is needed by hatire
struct  {
  int16_t  Begin;   // 2  Debut
  uint16_t Cpt;      // 2  Compteur trame or Code
  float    gyro[3];   // 12 [Y, P, R]    gyro (actually just the Y,P,R from the fusion filter)
  float    acc[3];    // 12 [x, y, z]    Acc (this is left empty and ignored)
  int16_t  End;      // 2  Fin
} hat;

/////////////////////////////////////////////////////////////////////////////////
// Prints filtered angles to either serial or BLE based on connection.         //
// Only runs in test mode and usb mode for now, hatire requires custom struct. //
//                          *FOR DEBUGGING*                                    //
/////////////////////////////////////////////////////////////////////////////////
void SerialPrint(char msg[]) {
  if (TestMode) {
    Serial.print(msg);
    /* if bluetooth mode
    if (BLEconnected) {
      hatireCharacteristic.writeValue(msg);
    }
    // else serial mode
    else {
      Serial.print(msg);
    } */
  }
}

///////////////////////////////////////////////////////////////////
// Sends hat struct to hatire using current communication method //
///////////////////////////////////////////////////////////////////
void sendAnglesToHatire() {
  // Send HAT  Frame to  PC base on communication mode
  // if bluetooth mode
  if (BLEconnected) {
    hatireCharacteristic.writeValue((byte*)&hat,30);
  }
  // else serial mode
  else {
    Serial.write((byte*)&hat,30);
  }
  hat.Cpt++;
  if (hat.Cpt>999) 
  {
    hat.Cpt=0;
  }
}

///////////////////////////////////////////////////////
// Updates the IMU and fusion filter to get new data //
///////////////////////////////////////////////////////
void updateAngles() {
  //  Check for new IMU data and update angles
  angles = imu.update();

  //  Wait for new sample - 7 ms delay provides a ~100Hz sample rate / loop frequency
  delay(6);
  loopCount++;

  if (TestMode){
    //  Display sensor data every displayPeriod, non-blocking.
    if (millis() - previousMillis >= displayPeriod) {
      Serial.print("Roll: ");
      Serial.print(angles.roll);
      Serial.print("\tPitch: ");
      Serial.print(angles.pitch);
      Serial.print("\tYaw: ");
      Serial.print(angles.yaw);
      Serial.print("\tLoop Frequency: ");
      Serial.print(loopFrequency);
      Serial.println(" Hz");
      loopFrequency = 0;
      previousMillis = millis();
    }
    loopFrequency++;
  }
  else {
    if (loopCount == -1) {
      center_yaw=angles.yaw;
      center_pitch=angles.pitch;
      center_roll=angles.roll;
    }

    //Assign yaw, pitch, and roll in hatire struct
    hat.gyro[0]=angles.yaw - center_yaw;
    //hat.gyro[1]=angles.pitch - center_pitch; //Roll axis disabled for now
    hat.gyro[2]=angles.roll - center_roll;

    // Send HAT  Frame to  PC
    sendAnglesToHatire();
  }
}

//////////////////////////////////
// Setup function, runs at boot //
//////////////////////////////////
void setup() {
  // Set hatire constants //
  // header frame
  hat.Begin=0xAAAA;
  // Frame Number or err code
  hat.Cpt=0;
  // footer frame
  hat.End=0x5555;

  // initialize the led digital pins as an output
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(RED, OUTPUT);
  pinMode(BLUE, OUTPUT);
  pinMode(GREEN, OUTPUT);
  pinMode(LED_PWR, OUTPUT);

  //Disable power LED until setup procedure is complete, Enable purple while booting up
  digitalWrite(LED_PWR, LOW);
  digitalWrite(RED, LOW);
  digitalWrite(GREEN, HIGH);
  digitalWrite(BLUE, LOW);
  digitalWrite(LED_BUILTIN, LOW);

  //Bluetooth Init
  BLE.begin();
  BLE.setLocalName("Nano 33 Head Tracker");
  BLE.setAdvertisedService(opentrackService);
  opentrackService.addCharacteristic(hatireCharacteristic);
  opentrackService.addCharacteristic(writeCharacteristic);
  BLE.addService(opentrackService);
  hatireCharacteristic.writeValue((byte*)&hat,30);
  BLE.advertise();

  //Serial Start
  Serial.begin(115200);

  //Initialise the LSM9DS1 IMU
  imu.begin();

  //Set filter settings
  imu.setFusionAlgorithm(SensorFusion::FUSION);
  imu.setFusionPeriod(0.01f);   // Estimated sample period = 0.01 s = 100 Hz
  imu.setFusionGain(10);       // Lower than 10 caused issues for me but you may be able to go as low as 5. 
                               // *Play around with gain if you have accuracy or slow response issues*
                               // Lower = More accurate but also more sluggish, Higher = Faster but less accurate.

  if (imu.connected()) {
    SerialPrint("LSM9DS1 IMU Connected.\n"); 

    //Start IMU
    imu.start();
  
    //Set boot up led RED only by disabling blue to indicate device is now running and waiting for connection
    digitalWrite(BLUE, HIGH);
  }
  else {
    // LSM9DS1 IMU not found
    digitalWrite(BLUE, HIGH);
    //Flash Red Light To indicate IMU error
    while(1) {
      digitalWrite(RED, LOW);
      delay(500);
      digitalWrite(RED, HIGH);
    }

  }
}

//////////////////////////////////////////////////////////////
// Attemps connection over BT and usb serial over and over. //
// Once connected, runs main loop until disconnected.       //
//////////////////////////////////////////////////////////////
void loop() {
  //LED red, pwr LED off while disconnected
  digitalWrite(RED, LOW);
  digitalWrite(LED_PWR, LOW);

  //Attempt Serial and Bluetooth connection
  BLEDevice central = BLE.central();
  delay(500);

  //if connected with BT or serial, else restart loop
  if (Serial || central); {

    //If bluetooth mode
    if (central.connected()) {
      BLEconnected = true;
      digitalWrite(RED, HIGH);
      digitalWrite(LED_PWR, HIGH);
      //call main loop, runs while connected
      while(central.connected()) {
        updateAngles();
      }
    }

    // Else usb serial mode
    else {
      BLEconnected = false;
      digitalWrite(RED, HIGH);
      digitalWrite(LED_PWR, HIGH);
      //call main loop, runs while connected
      while(Serial) {
        updateAngles();
      }
    }
  }

}
```

reefwing, 2023-01-30:
I probably wont get a chance to look at this until next week as I am in the middle of something else. 

Which version of the Nano 33 BLE are you using? The latest one is using a different IMU.

FugLong, 2023-01-30:
Thanks for the reply! 

> Yaw measurements have always been the least accurate in my experiments, you need to make sure that the magnetometer 
> has been compensated for soft and hard iron effects - I haven't explored this in a heap of depth (yet). 

This could be an issue. I saw some form of calibration in one of the fusion files and as it was already performing twice as well as madgwick and mahony I decided the calibration must be fine.

> Which version of the Nano 33 BLE are you using? The latest one has a different IMU.

Oh damn, you just blew my mind. I didn't realize there had been an update. I'm not sure which one I have. I just bought it direct from Arduino US like a week ago so it's safe to assume it has the new one. I have been treating it as an LSM9DS1 but I will verify and look into this further just to make sure.

I might get it figured out this week, but if I don't I would definitely appreciate any help you can offer. I'll keep this thread updated. I'll continue working on it and ask in the fusion discussions as well, thanks again.

reefwing, 2023-01-30:
Ha - this is the latest [Nano 33 BLE rev 2](https://store-usa.arduino.cc/products/nano-33-ble-sense-rev2-with-headers). 

I do plan to update the library to also support this version - I recently bought one but haven't played with it yet. Seems strange to change the sensors - maybe they can't get the LSM9DS1 anymore?

Yes please let me know if you find anything.

FugLong, 2023-01-31:
Ah I see now, so I do actually have the original IMU. I have the base 33 ble with only bluetooth and a LSM9DS1; I can't find a version 2 of it anywhere. The only version 2 boards are the sense model.

Good and bad news I guess. I'll focus in on fusion and see what could be going on

reefwing, 2023-02-04:
Hope you are feeling better!

They do seem a bit grumpy over at the Fusion repo 8-).

I'd be surprised if my NWU to NED conversion is related to your issue - I'm just changing the sign. I did this to match the outputs from the other algorithms. It is possible I'm using the wrong terminology. An easy way to test this is just comment out the conversion and use the output straight from the fusion algo.

I'll have a play around with it next week as well.

FugLong, 2023-02-06:
Thanks! Yeah they are pretty serious over there lol. 

You're right, the conversion isn't the issue for sure. I commented it out completely and the yaw is still affected by the others axis.

I did some extra delicate testing earlier and found that both other axis are actually affecting yaw, pitch just has a much stronger effect on it. Roll still throws it off more than it should but it's much more subtle. 

I ran all the way through the fusion update in reefwing today and honestly I can't see where the reefwing code could be breaking anything. You use the filter just like the advanced exampled demonstrates. I found an old forum post with someone having a similar issue with a totally different sensor and they just had to buy a new one... There may be some flaw in reefwing or fusion I don't understand but my nano 33 ble being busted is seeming more and more likely. 

I have simplified the code a bit so it's more readable and straight forward. If you can, do you mind flashing this and seeing if your yaw is affected by other axis too? 
```
#include <ArduinoBLE.h>
#include <ReefwingAHRS.h>

// LED Pin definitions
// *RGB PINS ARE INVERTED ON THIS BOARD*
// for RED, BLUE, GREEN, and BUILTIN only: HIGH means LOW and vice versa.
#define PIN_LED     (13u)
#define LED_BUILTIN PIN_LED
#define RED        (22u)
#define GREEN        (23u)
#define BLUE        (24u)
#define LED_PWR     (25u)

LSM9DS1 imu;
EulerAngles angles;

//TEST MODE ENABLE OR DISABLE
bool TestMode = true; //Set true to output data directly to serial (bypass hatire conversion + centering)

//Variables only required in test mode
int loopFrequency = 0;
const long displayPeriod = 100;
unsigned long previousMillis = 0;

//BLE variables + init
bool BLEconnected = false;
const char* deviceServiceUuid = "19b10000-e8f2-537e-4f6c-d104768a1214";
const char* deviceServiceCharacteristicUuid = "19b10001-e8f2-537e-4f6c-d104768a1214";
BLEService opentrackService(deviceServiceUuid); 
BLECharacteristic hatireCharacteristic(deviceServiceCharacteristicUuid, BLERead | BLEWrite | BLENotify, 30, true);

//this structure is needed by hatire
struct  {
  int16_t  Begin;   // 2  Debut
  uint16_t Cpt;      // 2  Compteur trame or Code
  float    gyro[3];   // 12 [Y, P, R]    gyro (actually just the Y,P,R from the fusion filter)
  float    acc[3];    // 12 [x, y, z]    Acc (this is left empty and ignored)
  int16_t  End;      // 2  Fin
} hat;

///////////////////////////////////////////////////////////////////
// Sends hat struct to hatire using current communication method //
///////////////////////////////////////////////////////////////////
void sendAnglesToHatire() {
  // Send HAT  Frame to  PC base on communication mode
  // if bluetooth mode
  if (BLEconnected) {
    hatireCharacteristic.writeValue((byte*)&hat,30);
  }
  // else serial mode
  else {
    Serial.write((byte*)&hat,30);
  }
  hat.Cpt++;
  if (hat.Cpt>999) 
  {
    hat.Cpt=0;
  }
}

[... 124 lines cut ...]
      digitalWrite(LED_PWR, HIGH);
      //call main loop, runs while connected
      while(central.connected()) {
        updateAngles();
        //  Wait for new sample - 7 ms delay provides a ~100Hz sample rate / loop frequency
        delay(7);
      }
    }

    // Else usb serial mode
    else {
      BLEconnected = false;
      digitalWrite(RED, HIGH);
      digitalWrite(LED_PWR, HIGH);
      //call main loop, runs while connected
      while(Serial) {
        updateAngles();
        //  Wait for new sample - 7 ms delay provides a ~100Hz sample rate / loop frequency
        delay(7);
      }
    }
  }

}
```

reefwing, 2023-02-06:
Give me a couple of days - I have 3 Nano 33 BLE's of various types that I can test. It is interesting that Arduino moved to a new IMU on the Sense - I wonder whether they have been having issues with the old one.

reefwing, 2023-02-11:
I decided to update the library to provide support for the new Nano 33 BLE Sense Rev. 2 and incorporate the latest version of Fusion which can now use the NED reference frame. Unfortunately, this turns out to be a large change. I need to separate the IMU library from the Sensor Fusion Algorithms. This will make it easier to support different hardware and IMUs in the future but is taking a bit of time. Once I've updated the Library I can get back to testing your Yaw issue.

FugLong, 2023-02-11:
Thanks for the update! I'll let you know if my new boards come in and I get to test myself before you finish those changes

reefwing, 2023-02-19:
FYI - The first step is done, I've pulled out the LPS22HB code and created a [seperate repository](https://github.com/Reefwing-Software/Reefwing-LPS22HB). Next up the IMU!

FugLong, 2023-06-30:
@reefwing Did you ever get a chance to test out my script to see if the axis are acting independently like they should? My package was lost in the mail then I got really busy and stepped away from my project for a while so I haven't even touched it since.

reefwing, 2023-07-01:
Hey @FugLong, no sorry. The update of the AHRS lib has turned into a mamoth project. I've separated out the LSM9DS1 components into a separate library and I've added a kalman filter option. I also decided to pull out the fusion option all together, since this library works very differently to the other sensor fusion algorithms I have. Plus xio updated their library so that it compiles for c++ now (I believe).

I'm in the final testing phase of the library, using a combination of different boards and IMUs. I'll keep an eye out for the phenomena that you are seeing. Note that pitch does effect yaw, if you are using one of the algorithms (CLASSIC, KALMAN, and NONE) that calls `tiltCompensatedYaw()`, then pitch (and roll) values will effect the pitch. The other algos use quaternions, but they also try to compensate for tilt in the yaw calculations.

```
void ReefwingAHRS::tiltCompensatedYaw() {
  //  Calculate yaw using magnetometer & derived roll and pitch
  float mag_x_compensated = _data.mx * cos(angles.pitchRadians) + _data.mz * sin(angles.pitchRadians);
  float mag_y_compensated = _data.mx * sin(angles.rollRadians) * sin(angles.pitchRadians) + _data.my * cos(angles.rollRadians) - _data.mz * sin(angles.rollRadians) * cos(angles.pitchRadians);

  angles.yawRadians = -atan2(mag_x_compensated, mag_y_compensated);
  angles.yaw =  angles.yawRadians * RAD_TO_DEG;    //  Yaw compensated for tilt
  //  float magYawAngle = atan2(_data.mx, _data.my) * RAD_TO_DEG;      //  Raw yaw from magnetometer, uncompensated for tilt - alternative yaw value
  
  angles.heading = angles.yaw - _declination;
}
```

FugLong, 2023-07-06:
Okay so my new arduino just arrived and it's the same story. Something with the fusion implementation I think, because while my mahoney version using reefwing is kind of broken, the axis do seem to be at least reacting independently. I'll see if I can get the other filters performing well enough for a real test to make sure

reefwing, 2023-07-08:
Ok that would explain why I'm not seeing it.

reefwing, 2023-08-08:
The latest version has just been released and I've pulled the Fusion algo out of it. On that basis I'm going to close this issue but let me know if I can help with anything else.

closed 2023-08-08 by reefwing