Reefwing-Software/Reefwing-AHRS #4: Modification to add 6dof (lsm3) and 9dof config split

FR86YoPPZ, 2023-02-16:
Hello.
I find your repository and it works very well for my nano 33ble sense.
I would like to use it on other mpu 9dof and also on a 6 dof sensor.
Could you split lsm9 config in specific .h and .cpp in aim to replace by other sensor?
Moreover could you add possibility to use 6dof (lsm3) sensor with less function(no bias etc)?
Thank you for your answer about possibility or problem.
If you don't have time could you give me different step to do this?
Thank you

reefwing, 2023-02-16:
Hi - glad you are enjoying the library.

Coincidently I am working on the very thing you have requested, i.e., separating out the sensors from the sensor fusion functions. I was prompted to do that in order to support the new Nano BLE 33 Sense Rev. 2 which uses two Bosch IMUs (BMI270 and BMM150) to replace the LSM9DS1.

I've started by pulling out the code for the LPS22HB pressure sensor and I have created a new improved library for it which includes altitude calculations and constant sampling (as opposed to one-shot readings). This new Library will be published by the end of next week.

I will then work on extracting the LSM9DS1 classes and creating a new library for that.

At the same time I will update the AHRS library so that the sensor readings can be passed in from whatever IMU you happen to use.

This library is targeted at the Nano 33 BLE family so I was going to autodetect which board is connected and add the appropriate sensor libraries, but that wont work for you unless I add the XIAO nRF52840 Sense to the mix. It looks like it uses the LSM6DS3TR-C IMU and is an interesting board. I will get one and see how hard it would be to incorporate support. No promises!

FR86YoPPZ, 2023-02-17:
Hi. Thank you very much for your quick answer and for the opportunity ;-)
I'm not enough skilled to try to add another sensor because all datas are load and get inside your library.
For me, the beast way could be as simple example "SensorFusion-1.0.5" library, it means that in main.cpp just :
#include "LSM9DS1.h" or "LSM3DS3.h"==>IMU (sensor library) / #include "SensorFusion.h", ax = IMU.acc.... and sensorfusion.MahonyUpdate(gx, gy, gz, ax, ay, az, mx, my, mz, deltat).. So it means have possibility to load datas from main.cpp and be easily to understand.
in your functions as LSM9DS1::complementaryUpdate(ax, ay, ....) and let possiblity to have 6DOF or 9DOF .
So i'm waiting with pleasure your update. thank you for your job !

reefwing, 2023-02-19:
FYI - The first step is done, I've pulled out the LPS22HB code and created a [seperate repository](https://github.com/Reefwing-Software/Reefwing-LPS22HB).

reefwing, 2023-08-08:
I've released the updated version which should work with just about any combination of Arduino Board and IMU. On that basis, I will be closing this issue. Any problems let me know.

closed 2023-08-08 by reefwing