Reefwing-Software/Reefwing-AHRS #5: Problem detecting the IMU

binosi, 2023-05-26:
Hi David,
first of all thank you very much for all your hard work, I have been maintaining few softwares myself and I know how much energy this can drain.

Now straight to the question (not sure it is the right place where to ask this but I will try, so feel free to ignore it; also full disclosure: I posted this on the Arduino forum but so far nobody replied). Just bought an Arduino Nano 33 BLE Sense Rev. 2 for a pet project I am doing with the kids.  I have installed the AHRS libraries and tried to run the testAndCalibrate sketch. It tells me "LSM9DS1 IMU not found". OTOH, if I use the standard Arduino_BMI270_BMM150 I can read data from the acellerometer, gyro and magnetometer. 

So at the moment I am a bit clueless on how to proceed: I have installed the libraries one at a time just to be sure that they do not conflict and found that I have failures of detecting the IMU also with other libraries like Femme Verbeeck's one. Am I missing something obvious?

Thanks for any spark in the dark you can provide.

reefwing, 2023-05-27:
Hi @binosi - thanks for your kind words.

The issue is that the Nano 33 BLE Rev1 and Rev2, use different IMUs. The testAndCalibrate example sketch is for the LSM9DS1 IMU, which you don't have.

You need something equivalent for the BMI270/BMM150.

Having said that, I should probably include an example on how to use the Rev 2 with my library. I'll have a look at that but it may take me a week or two due to existing committments.

binosi, 2023-05-27:
Ah crap, I thuoght the sensor was the same (as for the specs TL;DR, even though I teach my students a good dose of RTFM...).

To have an example on how to use your library with the Rev2 would be lovely, whenever it comes!
Thank you again and all the best!

reefwing, 2023-05-27:
Ha - reading the manual would take all the fun out of it!

There are actually quite a few changes that I need to make to my library to make it compatible with the new Rev 2 sensors, but I've started working on it. The up side is that it should make adding support for other boards easier as well. I'll post here when the updated library is uploaded.

I did also reply on the Arduino forum in case anyone else ends up on your thread.

binosi, 2023-05-29:
Thank you very much for this, David! 
I look forward seeing the new version of the libraries uploaded and start working with it.

reefwing, 2023-08-08:
The library has been updated and includes an example of how to use it with the Nano 33 BLE Rev 2 and the BMI270/BMM150 IMUs.

closed 2023-08-08 by reefwing