Reefwing-Software/Reefwing-AHRS #8: XIAO nRF52840 Lib Conflict

jonrichings, 2024-05-02:
Hi David,

Let me just say I am thrilled to find your corner of Github.  I've spent two days trying to find some libs and examples to use the IMU on the Xiao nRF52840 complete with fusion.  I am heading toward an air mouse kind of application and the Xiao seems to be ideal, but pulling the code together is a lot harder than I expected.

Anyway, I have stumbled with what I think is a conflict between your lib and Seeeduino\hardware\nrf52\1.1.8\cores\nRF5/wiring_constants.h:49
They both have #defines for RAD_TO_DEG.  This results in:
```
In file included from e:\Google Drive\My Documents\Arduino\libraries\ReefwingAHRS\src\ReefwingAHRS.cpp:34:
e:\Google Drive\My Documents\Arduino\libraries\ReefwingAHRS\src\ReefwingAHRS.h:45: warning: "RAD_TO_DEG" redefined
   45 | #define RAD_TO_DEG 57.2957795130823   //  180/PI * radians
      | 
In file included from C:\Users\Jon\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.8\cores\nRF5/Arduino.h:26,
                 from e:\Google Drive\My Documents\Arduino\libraries\ReefwingAHRS\src\ReefwingAHRS.cpp:30:
C:\Users\Jon\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.8\cores\nRF5/wiring_constants.h:50: note: this is the location of the previous definition
   50 | #define RAD_TO_DEG 57.295779513082320876798154814105
```

Obviously I want to use your lib, but the Seeed core is pretty important too.  Is this actually a bug or is there a stupid simple workaround that I don't understand?

Thanks for your help and your excellently documented libs.

Jon

reefwing, 2024-05-02:
Hi Jon,

Welcome! The problem is that both libraries are defining the same thing. You could either fork my library and change the name or probably the easier option is to import one library `undef` the macro and then import the next one. For example:

```
#include "ReefwingAHRS.h"
#undef RAD_TO_DEG
#include "SeeedLibrary.h"    // or whatever it is called
```

jonrichings, 2024-05-02:
Hi David, thanks for that.  Just tried the second option.  It doesn't seem to work.  Maybe because the conflicting lib is one of the core files and got "assimilated" first?  I tried reversing your suggestion so that the core file was replaced with the ReefWing one, but that didn't work either.    I'm going to try the forking your library option.  I've not done that before so wish me luck!

jonrichings, 2024-05-02:
OK, I forked your library, modified the names and got it working.  One thing though, I also had to manually edit:
```
#ifndef BOARD_NAME
#define BOARD_NAME "Seeed XIAO nRF52840 Sense"
#endif
```
Should BOARD_NAME have been automatically defined somehow?  
But, works for me and I'm all good now.  Thanks so much for your help :)

reefwing, 2024-05-02:
The library attempts to autodetect which board you have connected when you call `begin()`. The board types I have included and tested are:

```
enum class BoardType {
  NANO = 0,
  NANO33BLE,
  NANO33BLE_SENSE_R1,
  NANO33BLE_SENSE_R2,
  XIAO_SENSE,
  PORTENTA_H7,
  VIDOR_4000,
  NANO33IOT,
  NOT_DEFINED
};
```

So for you it should be returning `XIAO_SENSE`. What error were you receiving? Seeed may have changed the `BOARD_NAME` in their Arduino core library.

Most Arduino board variants defines a constant called `BOARD_NAME`, which you can examine using `Serial.println(BOARD_NAME);`

The `BOARD_NAME` of the XIAO should be `Seeed XIAO nRF52840 Sense`, and we use this to detect that the XIAO is attached. That is why your change works. Alternatively, we could check for the IMU on the I²C bus at address 0x6A.

Could you tell me what `BOARD_NAME` you are seeing (before you re-define it)? If it is different I will update the core library.

Thanks for highlighting this issue.

jonrichings, 2024-05-03:
If I don't put the board name in the #define, the error I get is:
```
E:\Google Drive\My Documents\Arduino\Reefwing_xiaoSense\ReefwingAHRS.cpp: In member function 'void ReefwingAHRS::begin()':
E:\Google Drive\My Documents\Arduino\Reefwing_xiaoSense\ReefwingAHRS.cpp:97:27: error: expected primary-expression before ',' token
   97 |     if (strncmp(BOARD_NAME, _boardTypeStr[4], 25) == 0) {
```

You have a line that already that says:
```
Serial.print("Detected Board - ");
  Serial.println(ahrs.getBoardTypeString());
```

And that returns: 
```
Detected Board - Seeed XIAO nRF52840 Sense
```

So as you would expect, so I am not sure why I have to #define it.

reefwing, 2024-05-04:
That's very weird. The error suggests that `BOARD_NAME` doesn't have a value. But if that is the case `getBoardTypeString()` shouldn't return the correct value. When I next update the library I will use the IMU address to identify the XIAO.

Anyway, glad you got it working!

closed 2024-06-18 by reefwing