Reefwing-Software/Reefwing-AHRS #12: Nano 33 BLE Rev2 yaw and heading not changing

ApeLerry, 2024-12-09:
Hi! I'm using the library with a Nano 33 BLE Rev2 board, the problem arises with the standard example code given for the board: the yaw and heading reading do not change in the right way even when rotating the board around the z axis. The only thing I changed in the code is the magnetic declination to match the one at my home. 
The Magnetometer is calibrated and it outputs correct readings (negative mx, positive my and mz). 

Moreover, the classic update method does not behave as it should: the roll, pitch and yaw readings are all NaN.

reefwing, 2024-12-09:
Hi @ApeLerry - can you confirm that you are using the latest version of the library? It should be v2.3.0. You can check what you have installed in the Arduino IDE library manager. The previous version had an issue where the CLASSIC filter could return NaN for some angles, but this was fixed in the latest release.

Regarding yaw - I have never had very good results from this for the Nano 33 BLE Rev 2. I suggest you calculate it yourself from the raw mag readings to see where exactly the issue is. Just copy the tilt compensated yaw function into your own code, then you can add some Serial.print()'s to see what is going on. If that doesn't work try the [Fusion](https://github.com/xioTechnologies/Fusion) library instead. This was written by Seb Madgwick.

```
void ReefwingAHRS::tiltCompensatedYaw() {
  //  Calculate yaw using magnetometer & derived roll and pitch
  float mag_x_compensated = _data.mx * cos(angles.pitchRadians) + _data.mz * sin(angles.pitchRadians);
  float mag_y_compensated = _data.mx * sin(angles.rollRadians) * sin(angles.pitchRadians) + _data.my * cos(angles.rollRadians) - _data.mz * sin(angles.rollRadians) * cos(angles.pitchRadians);

  angles.yawRadians = atan2(mag_y_compensated, mag_x_compensated);
  angles.yaw =  angles.yawRadians * RAD_TO_DEG;    //  Yaw compensated for tilt
  
  angles.heading = angles.yaw - _declination;
  angles.heading = fmod(angles.heading + 360.0, 360.0);  // Normalize to [0, 360)
}
```

ApeLerry, 2024-12-10:
Thank you so much! 
The NaN issue went away when I updated the library as you said, the yaw didn't change. 
It is a pity since I specifically bought the board with a magnetometer because the Rev1 had serious problems of gyro drift.
I will try what you suggested and let you know.

reefwing, 2024-12-10:
My pleasure. I'll close this issue but feel free to raise another one if something new comes up. Best of luck with your project.

closed 2024-12-10 by reefwing