Reefwing-Software/Reefwing-AHRS #13: Bug in ReefwingAHRS::madgwickUpdate function

martinbudden, 2024-12-29:
There's a bug in your implementation of the Madgwick update function at
https://github.com/Reefwing-Software/Reefwing-AHRS/blob/main/src/ReefwingAHRS.cpp#L570-L573

It seems like this is an inherited bug that is present in many implementations.

Mark Uckermann has done a thorough investigation at https://github.com/RideBeeline/madgwick-investigation/tree/main?tab=readme-ov-file#madgwicks-filter-implementation-analysis

Essentially the calculation of `bx` and `bz` is out by a factor of two.

The fix is at https://github.com/RideBeeline/madgwick-investigation/blob/main/bug_description.md#bug-2-in-the-c-version-2-missing

reefwing, 2024-12-29:
Hi @martinbudden - thanks for letting me know. I'll check it out and push through a fix.

reefwing, 2024-12-30:
Fixed in version 2.3.1

closed 2024-12-30 by reefwing

martinbudden, 2024-12-30:
Glad to be of assistance.