Reefwing-Software/Reefwing-AHRS #18: Example `boardType` fails to compile

iot49, 2025-01-08:
Example `boardType` fails to compile with the following error messages:

```cpp
/Arduino/libraries/Reefwing-AHRS/src/ReefwingAHRS.cpp: In member function 'void ReefwingAHRS::begin()':
/Arduino/libraries/Reefwing-AHRS/src/ReefwingAHRS.cpp:139:27: error: expected primary-expression before ',' token
  139 |     if (strncmp(BOARD_NAME, _boardTypeStr[4], 25) == 0) {
      |                           ^
/Arduino/libraries/Reefwing-AHRS/src/ReefwingAHRS.cpp: In member function 'void ReefwingAHRS::complementaryUpdate(SensorData, float)':
/Arduino/libraries/Reefwing-AHRS/src/ReefwingAHRS.cpp:489:25: error: '_d' was not declared in this scope; did you mean '_q'?
  489 |   const float qDot0 = - _d.gx * _att[1] - _d.gy * _att[2] - _d.gz * _att[3];
      |                         ^~
      |                         _q
/Arduino/libraries/Reefwing-AHRS/src/ReefwingAHRS.cpp:494:22: error: 'halfdT' was not declared in this scope; did you mean '_halfdT'?
  494 |   _att[0] += qDot0 * halfdT;
      |                      ^~~~~~
      |                      _halfdT

exit status 1

Compilation error: exit status 1
```

IDE:

```
Version: 2.3.4
Date: 2024-12-03T09:58:49.900Z
CLI Version: 1.1.1

Copyright © 2025 Arduino SA
```

reefwing, 2025-01-08:
Hi @iot49 - what board are you trying to compile on?

reefwing, 2025-01-09:
Well the library wont detect a custom board. The idea is it will assign the appropriate IMU and DOF based on known boards. If it can't identify the board it should default to `NOT_DEFINED`. The issue may be that you don't have `BOARD_NAME` defined. You can fix that by defining it yourself before calling the library, e.g.,

`#define BOARD_NAME "Custom ESP32"`

The other couple of errors you picked up are actually caused by a recent upgrade that was fixing a different issue. Corrected in version 2.3.5 which should push out to the Arduino Library Manager shortly.

I'm not sure what you mean about the sensor being tightly coupled, you can use any sensor you want as long as you pass the data in the expected format.

> Wouldn't some kind of "update" function on the ahrs library (with the new sensor values) make sense?

You mean like:

```
imu.updateSensorData();
ahrs.setData(imu.data);
ahrs.update();
```
??

iot49, 2025-01-09:
> You can fix that by defining it yourself before calling the library, e.g.,
>
> `#define BOARD_NAME "Custom ESP32"`

Just tried this: added the define before `#include <ReefwingAHRS.h>` in  the [boardType example](https://github.com/Reefwing-Software/Reefwing-AHRS/tree/main/examples/boardType). I still get the same errors.

> You mean like: ...

Something like this. The `ahrs` library would be separate and tied with particular imu's only via an api (e.g. like `setData` and `update`, as you show). Then there would be, also separate, libraries for various imu's. And possibly yet another library for board discovery. Basically, keep things that do not require tight coupling separate and interact via well defined and simple api's. This also helps testing and updates.

But of course that's just one way of doing things ...

closed 2025-01-09 by iot49