Reefwing-Software/Reefwing-AHRS #19: Setting Declination

cfox570, 2025-01-21:
What is the angle for the Magnetic Declination value?  I am located in the USA on the east coast. My value is 11 degrees 47' W.   Do I just use 47.783?  Or do I need to shift with 180 degrees?

reefwing, 2025-01-21:
Magnetic declination is the angle between true north and magnetic north, expressed in degrees east or west. To correctly use your magnetic declination value of 11° 47’ W (west), you need to convert it into decimal degrees.

Convert the minutes to decimal degrees:

- 	47'  is  47/60 = 0.783  degrees.
- 	Total magnetic declination:  11 + 0.783 = 11.783  degrees.

Since your magnetic declination is west, it is treated as negative. So, your value is -11.783 degrees.

`ahrs.setDeclination(-11.783);`

cfox570, 2025-01-22:
Thanks for the response.

closed 2025-01-22 by reefwing