Reefwing-Software/Reefwing-AHRS #20: Add Arduino Nano 33 BLE rev 2 version (not SENSE Rev2 version) as supported board in library

MCR68, 2025-02-09:
The library  ReefwingAHRS.cpp supports the Nano 33 BLE **SENSE** rev 2 only and not Nano 33 BLE rev 2 with  BMI270 and BMM150 IMUs too and is the same board without microphone and barometric pressure sensors.

....
ReefwingAHRS::ReefwingAHRS() { 
  _boardTypeStr[0] = "Nano";
  _boardTypeStr[1] = "Nano 33 BLE";
  _boardTypeStr[2] = "Nano 33 BLE Sense";
  _boardTypeStr[3] = "**Nano 33 BLE Sense Rev 2";**
  _boardTypeStr[4] = "Seeed XIAO nRF52840 Sense";
  _boardTypeStr[5] = "MKR Portenta H7";
  _boardTypeStr[6] = "MKR Vidor 4000";
  _boardTypeStr[7] = "Nano 33 IoT";
  _boardTypeStr[8] = "Undefined Board Type";
}

please add the case:

  **_boardTypeStr[x] = "Nano 33 BLE Rev 2";**

reefwing, 2025-02-09:
Hi @MCR68, I'm happy for you to add this as a PR if you want? Just follow what I did to differentiate the Sense R1 and R2. If you get stuck I can assist.

MCR68, 2025-02-10:
Thanks David for your fast reply.
I modified my local ReefwingAHRS.cpp forcing to load the BMI270_BMM150 IMUtype with the Nano 33 rev2 and all runs very well.
But sorry I am not familiar with how to add changes on Github to a project. this is my first post.

reefwing, 2025-02-11:
No worries. I've added support for the Nano 33 BLE Rev 2 in version 2.3.6 of the AHRS library. Should be available in a few hours from the Arduino library manager. 

How the detection code works is explained in a [Medium article](https://reefwing.medium.com/arduino-nano-33-ble-sense-rev-2-and-the-seeed-nrf52840-sense-3e16c5e25d95) which you might find interesting (or not 😂).

closed 2025-02-13 by reefwing